brandonlw/Psychson #81: Send Firmware on USB running HID Payload

sardoc, 2015-02-24:
So, after finally finding a USB with 2303, I followed demo 1 which worked great (didn't even need to short any pins).

So it works great with the Rubber Ducky payload, but now apparently all the PCs I am using don't recognize the device as a USB device (appears to be a keyboard according to device manager). I plug in the USB, the script runs, and a "Removable Disk (F:)" option appears, though double clicking on it just gives the "Please insert a disk into Removable Disk (F:)" error. Attempting to use GetInfo with DriveCom.exe results in a "System.InvalidOperationException: DeviceIoControl failed: 0079" error.

Before I take it apart and try to short the pins (if that would even help), just wanted to check to see if anyone else ran into this. And just to check, if I do have to short the pins, are the two pins to short the left-most pins the arrow points to in this pic: https://github.com/adamcaudill/Psychson/blob/master/docs/PinsToShortUponPlugInForBootMode.jpg

crappy44, 2015-02-24:
That is normal. If u short the Pins error 79 should be gone. It is Pin 2 and 3 in the Picture. Not Pin 1 and 2!

sardoc, 2015-02-24:
Thank you very much! Glad I asked about the pins first :)

That worked great to solve the problem, thanks again!

closed 2015-02-24 by sardoc