brandonlw/Psychson #83: Kingstone Datatraveler 100G3 32GB bricked!

zux88, 2015-03-05:
Hello,

I have 2 Kingstone Datatraveler 100G3 32GB. One of them started showing that there is no media on device and that he is 2303 PRAM device. So I tried copying firmware from my working USB to broken USB.
I dumped firmware from working device, and tried to send firmware to broken device.
The proces started, and after executing gave some IO error.
Now OS can't even load the device.
Is there any solution for this, or the device is bricked permanently?

zux88, 2015-03-06:
Issue solved.
At first I used a needle to put the device in boot mode, by connecting pin 2 and pin 3 on that Phison chip, then i wrote firmware with DriveCom utility. After a several tries I had a good firmware recognized by the OS, I used Restore v3.13 utility and my USB is now alive and working. Firmware and burner files were downloaded from russian site http://www.usbdev.ru, and there are a lot of utilities on http://flashboot.ru/. I hope somehow this helps, because your issues and comments helped me alot.

darf2k, 2015-06-22:
Very useful, thx.

minhtt159, 2015-12-01:
I have an Kingston DataTraveler 100G3 as well but it is a 16GB version. Can you send me a private message to give me some instant advise if something goes wrong?

soomeone, 2016-05-24:
Have the same problem, but a transmicro-mx stick. My stick's LED doesn't go on even I try to enter the boot mode...

zuhaib2002, 2016-09-18:
zux, if you would please explain how exactly did you turn it to boot mode, i am not to shorting pins and stuff so i have no idea how to practically do this, do u cut out the pin 2 and 3 or just attach them when plugged in with a conducting material?

zux88, 2016-09-19:
You should put a needle between two pins and then plug it in your pc.

Fabian9292, 2018-01-14:
Hi, i need help.... i have the same usb (G3 32GB). please tell me as it did.!!

Thx

romater0, 2021-06-09:
I have trouble finding the utilities from the russian site, could you please drop them here, so it'd be easier and the firmware as well