brandonlw/Psychson #85: SanDisk Ultra 16Gb USB 3.0 SDCZ48-016G problem

rger, 2015-03-18:
other pendrive : please help

Action specified: SendFirmware
Gathering information...
Reported chip type: 0000
Reported chip ID: 40-11-05-02-00-00
Reported firmware version: 254.07.00
Mode: Unknown
Switching to boot mode...
FATAL: System.InvalidOperationException: Header not accepted
   at DriveCom.PhisonDevice.TransferFile(Byte[] data, Byte header, Byte body) in
 c:\Users\Brandon\Documents\GitHub\PS2251-03\DriveCom\DriveCom\PhisonDevice.cs:l
ine 256
   at DriveCom.PhisonDevice.TransferFile(Byte[] data) in c:\Users\Brandon\Docume
nts\GitHub\PS2251-03\DriveCom\DriveCom\PhisonDevice.cs:line 238
   at DriveCom.Startup._ExecuteImage(String fileName) in c:\Users\Brandon\Docume
nts\GitHub\PS2251-03\DriveCom\DriveCom\Startup.cs:line 403
   at DriveCom.Startup._SendFirmware() in c:\Users\Brandon\Documents\GitHub\PS22
51-03\DriveCom\DriveCom\Startup.cs:line 375
   at DriveCom.Startup.Main(String[] args) in c:\Users\Brandon\Documents\GitHub\
PS2251-03\DriveCom\DriveCom\Startup.cs:line 114

sardoc, 2015-03-19:
Are you sure that's a Phison chip (since the chip type shows 0000, I am assuming it's not)? Again, the reported chip type needs to be 2303, or else the code won't work.

rger, 2015-03-19:
im check the version and program not show of this... flash drive information

MrDiagnose, 2016-10-24:
you need to manually swith your pendrive into bootmode by physically touching the two pins i know it coz i have the same pendrive as you ;)