brandonlw/Psychson #87: Building DriveCom etc

Arvandor, 2015-03-25:
I'm trying to make a BadUSB for use in testing our users against that kind of social engineering. The repository contains a bunch of Visual Studio solutions, but no built executables? I tried loading DriveCOM into my version of VS2012, but it didn't seem to like it.

How does one go about compiling or running these tools?

sardoc, 2015-03-25:
Opening the Psychson-master\DriveCom\DriveCom.sln file in Visual Studio 2012 (Ultimate) works fine over here - can you be more specific than "doesn't seem to like it"?

Once opened in VS2012, all you need to do is build it in order to get the DriveCom.exe. You'll also probably want to build the EmbedPayload solution as well.

Arvandor, 2015-03-25:
Aha, I think I just opened it wrong... I did new solution and opened the
top level folder for that project, and it complained about something (I
don't remember what.) Opening the .sln file seems to have worked. I've done
a little bit of C# and ASP.NET coding in Visual Studio, but I've never
tried to build out to an .exe, so just a bit noobish there haha.

I now can't find any documentation on how to use it to check the chip type
of the flash drive... It kicks me into console, and I've tried h, help, ?,
trying to figure out if there's any kind of built in documentation for what
is possible, but I don't see anything. The command is DriveCom.exe /drive=G
/action=?????

Thanks for the help.

sardoc, 2015-03-25:
You'll want to refer to this wiki page initially: https://github.com/adamcaudill/Psychson/wiki/Setting-Up-the-Environment

Once you've compiled the DriveCom.exe, you can just call it directly from a cmd window, rather than messing with it in VS2012. You'll want to run the GetInfo action initially to make sure you have the right type of chip (2303).

Once you found a USB you can use, you'll need a burner image - https://github.com/adamcaudill/Psychson/wiki/Obtaining-a-Burner-Image has information on this, and you can find links to locations where these images can be found.

Arvandor, 2015-03-26:
Yeah, I was running it in a command line, I just couldn't find the GeInfo
action. Thanks for the links, I'll follow those and see if I can harass you
as little as possible ;)

hackerpiyush, 2015-06-13:
Hey Arvandor can u plz send mee the compiled tools in zip format bcause my VB2012 iso file is permanently deleted so can u plz send me the compiled tool i will be very Thankful to u .... plz post soon

kernel-killer, 2015-10-30:
Hi, I have just now compiled the tools, heres the link: http://wikisend.com/download/695466/tools.zip

ykarli, 2017-02-27:
Hi
I just tried to make a usb hacker and I do not understand how to make drivecom.exe
may I ask you to send another folder tools?