brandonlw/Psychson #96: Kingstone DataTraveler Ultimate 3.0 G3 32 GB

donpiekarz, 2015-07-15:
Is supported the Kingstone DataTraveler Ultimate 3.0 G3 32 GB for Rubber ducky case?

Here is output from usbflashinfo:

```
Volume: E: 
Controller: Phison 2303 (2251-03) 
Possible Memory Chip(s):  
  Toshiba TH58TEG7DCK 
Memory Type: MLC 
Flash ID: 98D78493 72D7 
Chip F/W: 01.04.10 
Firmware Date: 2012-11-12 
ID_BLK Ver.: 1.2.23.0 
MP Ver.: MPALL v3.25.00 
VID: 0951 
PID: 1697 
Manufacturer: Kingston 
Product: DT Ultimate G3 
Query Vendor ID: Kingston 
Query Product ID: DT Ultimate G3   
Query Product Revision: PMAP 
Physical Disk Capacity: 31440502784 Bytes 
Windows Disk Capacity:  31436079104 Bytes 
Internal Tags: 2Q6P-S74J 
File System: UDF 
Relative Offset: 4032 KB 
USB Version: 3.00 
Declared Power: 896 mA 
ContMeas ID: 98DC-10-00 
Microsoft Windows 7 SP1 x64 
------------------------------------ 
http://www.antspec.com/usbflashinfo/ 
Program Version: 8.3.0.581 
```

Bonjour666, 2015-07-20:
Yes, it seems =) . Where did you buy it and when ?

donpiekarz, 2015-07-21:
I bought it 2 years ago in Poland, in some store with PC parts.

Bonjour666, 2015-07-21:
Ok, thanks

laMohsen, 2016-08-09:
@donpiekarz  any update cause i have the same one and need to know how it went with it?