brandonlw/Psychson #99: Toshiba 16GB USB 3.0 Flash Drive TransMemory MX (03-2251) broken after HID demo

zodiacGer, 2015-08-08:
Hey,

i know this issue has been reported several times now, but I still don't get why this error happens. 

I succeeded with the HID demo, and I can still enter bootmode with drivecom, but i can't update the firmware anymore. I keep getting this error, I hope you can help me!

Action specified: SendExecutable
FATAL: System.InvalidOperationException: DeviceIoControl failed: 0079
   bei DriveCom.PhisonDevice._SendCommand(SafeFileHandle handle, Byte[] cmd, Byte[] data, Int32 bytesExpected) in c:\Users\Cr0wTom\Desktop\Psychson-master\DriveCom\DriveCom\PhisonDevice.cs:Zeile 365.
   bei DriveCom.PhisonDevice.SendCommand(Byte[] cmd, Int32 bytesExpected) in c:\Users\Cr0wTom\Desktop\Psychson-master\DriveCom\DriveCom\PhisonDevice.cs:Zeile 295.
   bei DriveCom.PhisonDevice.TransferFile(Byte[] data, Byte header, Byte body) in c:\Users\Cr0wTom\Desktop\Psychson-master\DriveCom\DriveCom\PhisonDevice.cs:Zeile 249.
   bei DriveCom.PhisonDevice.TransferFile(Byte[] data) in c:\Users\Cr0wTom\Desktop\Psychson-master\DriveCom\DriveCom\PhisonDevice.cs:Zeile 238.
   bei DriveCom.Startup._ExecuteImage(String fileName) in c:\Users\Cr0wTom\Desktop\Psychson-master\DriveCom\DriveCom\Startup.cs:Zeile 403.
   bei DriveCom.Startup.Main(String[] args) in c:\Users\Cr0wTom\Desktop\Psychson-master\DriveCom\DriveCom\Startup.cs:Zeile 109.

zodiacGer, 2015-08-08:
okay i short the right pins^^ that beginner luck was on my side.

theSoenke, 2015-11-01:
@zodiacGer Do you mean this pins? I tried it, but it didn't work

[image]

zodiacGer, 2015-11-01:
Hi, i made a video for a friend few days ago: https://m.youtube.com/watch?v=rIyhs6-7OrE

; )

theSoenke, 2015-11-01:
It worked! Thanks a lot :D