brandonlw/Psychson #109: PS2251-61 BURNER IMAGE

kar2013, 2015-10-17:
HI,

Sombody can help to find the burner image of the microcontroller phison PS2251-61
i'don't unterstant what is the "yyy" version number

Thanks for your help

kar2013, 2015-10-18:
it's ok i find it !

http://flashboot.ru/files/file/172/

Thxs

xeanarth, 2015-10-20:
are u trying if it work on that version?

kar2013, 2015-10-20:
Hi xeanarth,

yes I try it, but this version don't work on this microcontroller !

xeanarth, 2015-10-20:
Do you have any screen or stamp about the error? or do u remember it? It's say to me that he cant load the Header in the phisonDevice adn startup c# files ! anything like to you too?
thx for the help :)

kar2013, 2015-10-20:
i have the same error, the processing stop in the header, 
Sytem can't flash,  microcontroller is not compatible with   this exploit, you need to edit this code !
Big challenge !

xeanarth, 2015-10-21:
Do you know anything in details?

kar2013, 2015-10-21:
No it's different for any device,  you need skills in microelectronic, ASM, C, C#
I am too light !