brandonlw/Psychson #111: Phison 2251-07

B00sterman, 2015-10-19:
Hi,
I know this has been posted multiple times, but I would like to test this vulnerability, whether with this script or not. I've got a Patriot Supersonic Xpress 16 GB, and, of course, I was gifted with the 2251-07 version. Is there ANY possibility to get this working on the 07?

xeanarth, 2015-10-20:
i follow, same as you need that info :)

B00sterman, 2015-10-20:
Yep :/ Doesn't seem like it tho. I am familiar with C++ etc, but only application coding. I don't know much about binaries applying to other chip types, like the 8051 one used here...

xeanarth, 2015-10-20:
can we have a private talk? my mail is leox.howl@gmail.com . 
Send me a mail or share your here :D

Isitha, 2015-12-05:
Same here it sucks, I wish there was an update. I could help as well, I know a bit of c++

haaha123123123, 2016-02-09:
selling 2251-03/2303 chips flash disk,contact me if you want buy
I have lots of 2251-03/2303 flsh disk,and it works with badusb,if anyone want to buy,please contact me, my email is haaha123123@126.com

kim-yannick, 2016-02-09:
GitHub is not a selling platform. Advertise your offer/scam somewhere other
like eBay.