brandonlw/Psychson #112: Rubber ducky Normal USB

RedToor, 2015-10-20:
Hi Guys, i have a question.
I can do this with Modem-3g usb (mobile)???

Thanks

or just USB 3.0 

Patriot 8GB Supersonic Xpress*
Kingston DataTraveler 3.0 T111 8GB
Silicon power marvel M60 64GB
Patriot Stellar 64 Gb Phison
Toshiba TransMemory-MX USB 3.0 16GB
Toshiba TransMemory-MX USB 3.0 8GB
Kingston DataTraveler G4 64 GB
Patriot PSF16GXPUSB Supersonic Xpress 16GB
Silicon Power 32GB Blaze B30 (SP032GBUF3B30V1K)
Kingston Digital 8GB USB 3.0 DataTraveler (DT100G3/8GB)

kar2013, 2015-10-20:
you need to know the exact microcontroller !
Use usbinfo or other !

RedToor, 2015-10-21:
Ok, i'll try it

zhymer96, 2016-06-01:
Hello that such I have a USB 3.0 DataTraveler 100 g3 , someone already tried this model ??