brandonlw/Psychson #113: Phison 2251-07 Unsolder & Solder a new

kar2013, 2015-10-21:
Hi all,

Do you think, it's possible to unsolder le micorcontroller  on device with original chip like 2251-XX
and solder a new compatible with this exploit ?

http://www.ebay.fr/itm/genuine-NEW-PHISON-PS2231-chip-LQFP64-/181894366869?hash=item2a59bfc695:g:z-QAAOxyyq5TNByF

Thanks for your answers !

apocsantos, 2015-10-21:
Possible it is, but you will need some equipment, cause for such tinny chip
soldering is quite hard with a standard iron.

António Pedro Oliveira Cunha Santos

*"Patience is the companion of wisdom." Aurelius Augustinus Hipponesis"*

*Contribua para a defesa do ambiente. Não imprima este e-mail se não for
necessário. *

*Please consider the environment before printing this e-mail.*

Aviso de Confidencialidade: Este e-mail e quaisquer ficheiros informáticos
com ele transmitidos são confidenciais, podem conter informação
privilegiada e destinam-se ao conhecimento e uso exclusivo da pessoa ou
entidade a quem são dirigidos, não podendo o conteúdo dos mesmos ser
alterado. Caso tenha recebido este e-mail indevidamente, queira informar de
imediato o remetente e proceder à destruição da mensagem e de eventuais
cópias.
É estritamente proibido o uso, a distribuição, a cópia, total ou parcial,
ou qualquer forma de disseminação não autorizada deste e-mail e de
quaisquer ficheiros nele contidos. O correio electrónico não garante a
confidencialidade dos conteúdos das mensagens. Caso o destinatário deste
e-mail tenha qualquer objecção à utilização deste meio deverá contactar de
imediato o remetente.

kar2013, 2015-10-21:
i have all the equipment : Irda station, hot gun air, flux...