brandonlw/Psychson #121: 2251-67 still no support?

dummys, 2015-12-01:
Hi guys,  for a forensic project I need to access secure hidden partition from a flash with this controller. Did you have a solution for it now? Can you help to try to patch it or create a custom firmware?  Do you think I can send the special unlock command thru usb or I will need the password as well,  therefore only patching will work? 
Thanks for your answer.

dummys, 2015-12-12:
thanks for such greate answer..............

davodamirajam, 2016-01-04:
Hi I am work in goldiran lg company
We also make flash
Working with phison,SMI,alcor
I have some firmware 
Example fw67v6.10.53
Fw67v6.11.10
If some one need email me

davodamirajam, 2016-01-04:
In warranty we first clean flash memory erase with uptool
And use mpall+new firmware 
I also share files and how to extract firmware 
In flash boot

davodamirajam, 2016-01-04:
http://flashboot.ru/flash_recovery/2015/05/03/extract-2251-03-firmware.html

davodamirajam, 2016-01-04:
I see some body say after firmware sail killed flash. I say no before insert to PC hold nand chip 29+30 with one piece as wire hold for 3second and let it 
In hardware boot mode you must hold 2 pin io

dummys, 2016-02-01:
davodamirajam Are you able to dump the firmware ?

davodamirajam, 2016-02-04:
let me to example something

for example we have a 2251-07 98 de 98 92

if i dump firmware and write to another flash disk

if second flash have a hardware bad block ???

so firmware is crashed

before the dump and write must be re mark bad block

closed 2017-03-21 by dummys