brandonlw/Psychson #124: DeviceIoControl: failed 0006

dickulus, 2016-01-03:
Hello,
i bought Patriot Supersonic Xpress 8GB (USB 3.0), but i am unable to dumb or flash any firmware. GetInfo says it has 2307 controller, but it is listed on supported devices page. I am able to set it into boot mode, but after it, i always get this error if i try to flash/dumb:
System.InvalidOperationException: DeviceIoControl failed: 0006
Is this device incompatible or am i doing something wrong?
Thanks

CleverShinobi, 2016-01-03:
Not compatible. Needs to be 2303

dickulus, 2016-01-04:
but on supported devices page, there is written that it is compatible too

Makuros, 2016-01-10:
http://ge.tt/4lL84YV2/v/0 download this and run the file. If the controller version is not 3, it means it's incompatible.