brandonlw/Psychson #136: Adding Support for other USB controllers.

abdouretro, 2016-04-12:
Why is the focus only on 1 Phison controller when other models already have their firmwares leaked.
SRLab is maintaining a page about the vulnerable devices which I believe should be targeted.
I believe the most Important thing here is the no boot patch to permanently protect the drives from tampering. Flashboot and USBDev are great places to start building a repo of original firmwares. One which I can be in charge of or contribute to a lot.

Thanks,
Awaiting a timely response.

Other similar issues :

Dude trying 2251-33 , https://github.com/adamcaudill/Psychson/issues/95
Dude posts info about 2303 , https://github.com/adamcaudill/Psychson/issues/79
2251-06 support request , https://github.com/adamcaudill/Psychson/issues/78
seems like useless similar issue , https://github.com/adamcaudill/Psychson/issues/71

Raycer, 2016-04-13:
#137

dk000000, 2016-04-14:
You say that "USBDev are great places to start building a repo of original firmwares"

How can you be so sure that the firmware data base on usbdev does not contain malicious code within it, or if it is even original?

I know it works, I've tested it.  But now I have no idea if my flashed USB stick is executing commands that I don't know about, so I must trash it.

dk000000, 2016-04-14:
On a side note, it is very scary thing that these masses of (possibly malicious) firmware are out available on the market, while (aside from HMP.A) there is zero effective mitigation software available to combat badusb intrusions.  Just consider how many used thumb drives are sold online daily....

dk000000, 2016-04-14:
forums.sandisk.com/t5/All-SanDisk-USB-Flash-Drives/sandisk-cruzer-8gb-firmware-1-27/td-p/331577

abdouretro, 2016-04-15:
Well, we have to start somewhere.. We're backed into a corner here, what's our other option, toss out all usb devices.. We'd still have the webcam and sd reader in the laptop btw.. Or write firmware from scratch without documentation.. I can barely read and write python.. I doubt anybody can do that to hegin with.. A simple repo to start then some reversing to try and find discrepancies.. A good way would be to compare different versions of firmware's code. Let's roll the dice on the possibility that not all firmwares can be poisoned and maybe..just maybe..a vendor could cooperate, or even better give us and official firmware or commands to dump the current firmware.. If we can dump firmware from factory sticks ..we can for example see if version 1.26 always has the same hash.. One of them has to be clean.. Unless they were made malicious from the beginning.

Also, know that karsten nohl revealed 2 vectors of attack.. If we can make sure the sticks don't do these signs.. We can build some confidence. After all.. You said it.. Zero mitigation.

Kira-Fs, 2022-08-30:
Phison 2303-70 supported??