brandonlw/Psychson #137: Some guy seems to have gotten it working on 2307 chip

Raycer, 2016-04-13:
https://translate.google.com/translate?hl=es&sl=ja&tl=en&u=http%3A%2F%2Fvivibit.net%2Fpsychson%2F

Seems like some japanese guy got it working, i really don't get how but maybe a dev will. 
I would love to colaborate, and i do have ASM and C, C++ knowledge, reversing tho is not one of my fortes.

Medusalix, 2016-08-26:
The author has published a guide, where here explains in detail how he got it working: https://translate.google.com/translate?sl=auto&tl=en&js=y&prev=_t&hl=en&ie=UTF-8&u=http%3A%2F%2Fvivibit.net%2Fpsychson2307%2F&edit-text=&act=url

Maybe someone can try it out :grinning:.

zerozyx, 2016-09-08:
any updates on this?

FoxSauerkraut, 2016-09-11:
Yeah, think I almost got it working.
It took so much time to get there, I will continue tomorrow.
At this point I was able to dump the sticks firmware, and rewrite a custom one.
Right now I cant tell for sure if the payload works, cause there are still some issues.

mijailr, 2016-09-21:
@FoxSauerkraut do you have any update? Did it work?

FoxSauerkraut, 2016-09-23:
Yes, I do have an update on this:
Flashing procedure seems to work as it should, but I still got major issues to consistently execute the payload on targets pc.
I'm going to write down what I did, but for now it needs some more work to be done.

Raycer, 2016-09-30:
@FoxSauerkraut Yes! thanks man, it's been a while since anybody did real work on this. If you can use any help i am willing if i can.

Please let us know when you publish that paper.

Killberty, 2016-10-06:
@FoxSauerkraut i have a device with a 2307 chip which i have no use for, so if oyu need someone to test i am willing to help :)

m4ccbr, 2016-11-05:
@FoxSauerkraut  I'm trying to do it by myself, but I have no clue how you execute step 4

patch -p0 < firmware-firmware2307.diff
patch -p0 < DriveCom-DriveCom2307.diff

darklite404, 2016-11-30:
@m4ccbr use linux command or manual edit 
- mean delete that line
+ mean add that line

I try this until the step that I successful build fw.bin
but after this step I fail.

Nathansswell, 2017-01-10:
@m4ccbr I have found the japanese guy's tutorial video on how it's done. http://vivibit.net/how_to_apply_psychson2307patch/ might want to watch it at half speed.

Make sure you run cmd as admin, otherwise it will not work.

I am currently working on a 2307 to see if I can make it work. I think I'll write a tutorial that better translates to English though. Hopefully I can manage this. I don't wanna brick the flash drive.

Pastitas, 2017-03-01:
any news on that @Nathansswell

Nathansswell, 2017-03-01:
I was really trying hard for like a week to understand how to do it. There are some bit of data that need changed within the .bin and I was very confused on which bits it was and how it was supposed to be done. That is where I stopped. I kept looking and trying different things, still have no idea. Maybe I should ask the guy again.

xxx2zzz, 2017-04-02:
I wrote the complete description in incomplete English.
http://vivibit.net/psychson2307final-en/

J0rdyV, 2017-05-01:
Didn't work for me,
Maybe you guys could help? :)

The DriveCom.exe gives this output...

_C:\Users\jordy\Desktop\Psychson-master\tools>DriveCom.exe /drive=D /action=SendF
irmware /burner="Burner\BN07V502TAW.BIN" /firmeware="Firmware\fw.bin"
Action specified: SendFirmware
Gathering information...
Reported chip type: 2307
Reported chip ID: 00-98-9E-94-92-76
Reported firmware version: 5.02.41
Mode: Burner
FATAL: System.ArgumentNullException: Pad kan niet null zijn.
Parameternaam: path
   bij System.IO.FileStream.Init(String path, FileMode mode, FileAccess access,
Int32 rights, Boolean useRights, FileShare share, Int32 bufferSize, FileOptions
options, SECURITY_ATTRIBUTES secAttrs, String msgPath, Boolean bFromProxy, Boole
an useLongPath, Boolean checkHost)
   bij System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access,
 FileShare share, Int32 bufferSize, FileOptions options, String msgPath, Boolean
 bFromProxy)
   bij System.IO.FileStream..ctor(String path, FileMode mode)
   bij DriveCom.Startup._RunFirmware(String fileName) in c:\Users\jordy\Desktop\
Psychson-master\DriveCom\DriveCom\Startup.cs:regel 413
   bij DriveCom.Startup._SendFirmware() in c:\Users\jordy\Desktop\Psychson-maste
r\DriveCom\DriveCom\Startup.cs:regel 378
   bij DriveCom.Startup.Main(String[] args) in c:\Users\jordy\Desktop\Psychson-m
aster\DriveCom\DriveCom\Startup.cs:regel 114_

Sorry that some words are in dutch 

Also tried PsychsonGUI, 
Worked first time but when i plug in the usb nothing happens,
When i did it the second time it said 'Header not accepted' when i try to flash the firmware.

Medusalix, 2017-05-01:
I think your parameter "/firmeware" should actually be written "/firmware" 😉.

J0rdyV, 2017-05-01:
Thx :p 

Now it says this ...
The same error as in PsychsonGUI : 'Header not accepted'

C:\Users\jordy\Desktop\Psychson-master\tools>DriveCom.exe /drive=D /action=SendF
irmware /burner=Burner\BN07V502TAW.BIN /firmware=Firmware\fw.bin
Action specified: SendFirmware
Gathering information...
Reported chip type: 2307
Reported chip ID: 98-DE-94-93-76-51
Reported firmware version: 1.05.10
Mode: BootMode
Rebooting...
Sending firmware...
FATAL: System.InvalidOperationException: Header not accepted
   bij DriveCom.PhisonDevice.TransferFile(Byte[] data, Byte header, Byte body) i
n c:\Users\jordy\Desktop\Psychson-master\DriveCom\DriveCom\PhisonDevice.cs:regel 256
   bij DriveCom.Startup._RunFirmware(String fileName) in c:\Users\jordy\Desktop\
Psychson-master\DriveCom\DriveCom\Startup.cs:regel 427
   bij DriveCom.Startup._SendFirmware() in c:\Users\jordy\Desktop\Psychson-master\DriveCom\DriveCom\Startup.cs:regel 378
   bij DriveCom.Startup.Main(String[] args) in c:\Users\jordy\Desktop\Psychson-m
aster\DriveCom\DriveCom\Startup.cs:regel 114

closed 2022-11-23 by Pastitas