brandonlw/Psychson #140: This one could work ?

lagar, 2016-06-03:
Phillips USB 64GB
Volume: D:
Controller: Phison 2303 (2251-03)
Possible Memory Chip(s): 
  Toshiba TH58NVG9T2JTA20
  Toshiba TH58TEG9T2JTA20
Memory Type: TLC
Flash ID: 983CA992 7AD7
Chip F/W: 01.07.53
Firmware Date: 2013-06-21
ID_BLK Ver.: 1.2.47.0
MP Ver.: MPALL v3.70.0E
VID: 13FE
PID: 5200
Manufacturer: Philips
Product: USB Flash Drive
Query Vendor ID: Philips 
Query Product ID: USB Flash Drive 
Query Product Revision: PMAP
Physical Disk Capacity: 61918150656 Bytes
Windows Disk Capacity:  61914017792 Bytes
Internal Tags: 2Q2P-S78J
File System: NTFS
Relative Offset: 4032 KB
USB Version: 3.00
Declared Power: 504 mA
ContMeas ID: E9A2-06-00
Microsoft Windows 7 SP1 x64 Build 76100
------------------------------------
http://www.antspec.com/usbflashinfo/
Program Version: 8.5.0.590

DanielL99, 2016-06-03:
I think so, because it has the right chip
Where do you bought it?

soomeone, 2016-06-03:
Has the right controller -> Yeah, it should

bsfoolery, 2016-06-03:
Volume: G:
Controller: SanDisk
Possible Memory Chip(s): Not available
VID: 0781
PID: 5567
Manufacturer: SanDisk
Product: Cruzer Blade
Query Vendor ID: SanDisk 
Query Product ID: Cruzer Blade    
Query Product Revision: 1.27
Physical Disk Capacity: 4004511744 Bytes
Windows Disk Capacity:  4002529280 Bytes
Internal Tags: 3Y4F-8CB2
File System: FAT32
Relative Offset: 16 KB
USB Version: 2.00
Declared Power: 200 mA
ContMeas ID: D91A-01-00
Microsoft Windows 7 SP1 x64 Build 7601
------------------------------------
http://www.antspec.com/usbflashinfo/
Program Version: 8.5.0.590

and this it's work?

bsfoolery, 2016-06-03:
i think this its work look the controller
--------------------------------------------
Volume: G:
Controller: Phison 2251-32 (2232)
Possible Memory Chip(s): 
  Micron MT29F16G08MAA
Memory Type: MLC
Flash ID: 2CD5943E 74
Chip F/W: 01.05.10
Firmware Date: 2008-01-26
VID: 0951
PID: 1607
Manufacturer: Kingston
Product: DataTraveler 2.0
Query Vendor ID: Kingston
Query Product ID: DataTraveler 2.0
Query Product Revision: PMAP
Physical Disk Capacity: 2006974464 Bytes
Windows Disk Capacity:  2002583552 Bytes
Internal Tags: 2QYP-S992
File System: FAT
Relative Offset: 4032 KB
USB Version: 2.00
Declared Power: 200 mA
ContMeas ID: D91A-03-00
Microsoft Windows 7 SP1 x64 Build 7601
------------------------------------

TotalDay, 2016-08-21:
Sandisk SDCZ48-016G

PS2251-03

https://goo.gl/88ASp7

Toshiba V3SZK-032G

PS2251-03

https://goo.gl/L0EK7u