brandonlw/Psychson #141: Psychson with GUI

soomeone, 2016-06-03:
This is not an issue, but i didn't found any other way to make a comment or communicate with brandonlw.

I created a fully gui programm (based on psychson) that let you flash your stick with only a few clicks. It includes already every tool needed to make a custom FW; but you have never to get in touch with them, because PsychsonGUI makes all the work for you :)

[image]

Finally, here the link to the Github page: [https://github.com/soomeone/PsychsonMaker/wiki](url)

tim241, 2016-06-12:
Cool but you should add a port feature thatwill automaticly port it to your USb controller.

soomeone, 2016-06-12:
I recently worked only with the Phison 2251-03, so I have no knowledge about the process on other controller. But if you need flashing a specific controller, post the name of it and I'll look if it will be possible to add it.

tim241, 2016-06-13:
ok I am working on a duckyscript to visual basic script converter so I think I don't need it anymore. If you want to help me. Then just mail me: tim241@ziggo.nl

tim241, 2016-06-23:
Can you maybe try to add 2307 support? I am going to buy a USB that will may ship with 2303 or 2307. Your tool is great I am going to use it much if I have my USB in home.

soomeone, 2016-06-23:
It is very unlikely that this will be possible with the 2303 controller, but i'll do some researches on it. 
If there are other requests or issues, please post them on the tool's github site.

tim241, 2016-06-23:
okejj :-)

H8to, 2016-12-23:
There are information about a successful port to 2307 on a japanese blog:

http://vivibit.net/psychson/
http://vivibit.net/psychson2/
http://vivibit.net/psychson2307/

They contain quite a lot of information. If any one wants to give it a try maybe it can be integrated into this tool.