brandonlw/Psychson #144: 2301 possible compatibility?

judge2020, 2016-07-31:
So I recently bought the recommended supersonic express 8gb flash drive for badUSB operations, but to my surprise instead of getting the expected 2307 clip or getting lucky with 2303 I am awareded a 2301 firmware. Since this is a chipset before 2303 I wondered if the exploit will work on it. 

`SetBootMode` and `DumpFirmware` drivecom modes work perfectly, the only thing that doesn't work is `SendFirmware`.
 I've tried the burner image for the 2301 as well as all 2303 ones, all returning `FATAL: System.InvalidOperationException: DeviceIoControl failed: 0079`.

TotalDay, 2016-08-23:
Sandisk SDCZ48-016G

PS2251-03

https://goo.gl/88ASp7

Toshiba V3SZK-032G

PS2251-03

https://goo.gl/L0EK7u

TheDarkSidesMinion, 2016-11-20:
I know its old but i thought i will comment. You can flash firmware with the Phison MPALL tool