brandonlw/Psychson #145: What the actual hell?

NovaHowl, 2016-08-08:
Alright, I'm pretty pissed now.

I was searching my house for usbs and I found a 2303 (Sandisk Ultra) and tried flashing it. After doing so, Windows stopped detecting one of my HDD's. The USB got flashed in some way but Windows gives an error (Could not recognize device. It may have malfunctioned). Where the hell is my HDD? I checked the SATA connections, all was fine. Checked the PSU connections. All Fine. How do I get my Hard Drive to show? My main SSD (OS) and my secondary SSD is fine but my Hard Drive is not showing up. Please help.

NovaHowl, 2016-08-08:
EDIT: I hid my System Reserved in disk managment so It wouldn't looked cluttered, and I think I may have messed up that. Shit, how do I fix it? I don't care if I lose data on that drive, just please help me.

reoogieste, 2016-08-08:
Is the HDD showing up in Disk Management?

NovaHowl, 2016-08-08:
Well I it shows 3 drives with no names. Two which I cannot do anything and the third I can right click but everything's greyed out. I guess it's my fault for not seeing this :(

NovaHowl, 2016-08-08:
[image]

reoogieste, 2016-08-09:
How big was your HDD ? It probably just needs to be re-formatted.

NovaHowl, 2016-08-09:
Woops, ignore the close issue thing.
Anyways, the HDD is 500 GB

NovaHowl, 2016-08-09:
I also managed to assign only one system reserved a drive letter. The C: drive. The rest I think is just screwed.

NovaHowl, 2016-08-12:
Plugged it in a docking station, copying all files. I'll format it after, plug it into a SATA port and see if Windows detects the drive.

NovaHowl, 2016-08-12:
Nope, not working. Maybe it's a dead SATA port. I'm going to try plug it in another one.

NovaHowl, 2016-08-12:
*Faceplant* I'm so sorry, the SATA cable to the Mobo got disconnected. I feel so dumb ;(

closed 2016-08-12 by NovaHowl