brandonlw/Psychson #149: I think its time to move on.

zuhaib2002, 2016-09-07:
The all-famous 2303 Controller is no longer easily available in the market.
Does it mean that this "Flaw" has been fixed in the more common 2307?
No one in 2 years have worked on Phison 2307 controllers and i doubt if someone is going to.
Conclusion : BADUSB vulnerability has been fixed

brandonlw, 2016-09-07:
"BadUSB" is not a vulnerability -- it is the ability to alter the firmware on a USB flash drive, and that most certainly has not changed on the 2307.

What prevents people from easily using this on the 2307 (and even some 2303s) is the patches, which need to be tweaked to work properly on different firmware versions. Rather than simply hard-code patch addresses, the code in this repository tries to be a tiny bit smart and look for code patterns, so that there'd be some hope of it working on other firmware versions, but it does not work well enough on the 2307 or some 2303 versions.

Studying the disassemblies and making the appropriate changes is all that's necessary -- and I've seen links on this very issue tracker (such as https://github.com/brandonlw/Psychson/issues/137 ) of people who have done exactly that, so it is certainly possible.

zuhaib2002, 2016-09-08:
Thank You for finding time to reply.
i never said it is not possible, but no one is committed enough to work further on this and more importantly on other controllers. 
only if the developers who discovered this 'ability' could find time for this project and somehow tweak through the changes for 'further analysis and research'

zuhaib2002, 2016-09-08:
Also i went thoroughly on the Japanese guy who sorted this out, but he is not actually providing documentation or the proper walk-through of how to do this on 2307