brandonlw/Psychson #152: I think we need a new manager

AshtinBlanchard, 2016-09-11:
I think it is time that we, the people, take charge and fork the repository and popularize the forked edition as we can work on it as a team rather than having this repo, which appears to be dead going forward. I have created a fork of this repo and am willing to add anyone willing to work on the project, I just ask for responsibility and consideration. Anyone wanting to collaborate on this can leave a comment saying they do. Feedback on my idea is welcomed with open arms, I'm not afraid of criticism.
[The forked repo](https://github.com/ashtinblanchard/Psychson/)

Related issues: 
#4 #98 #150 #149 #146 #136 #95 #78 #71 and many more, I assume. - Shows that the microcontroller this firmware works on is outdated.
#137 - Shows it is possible to get it working on other microcontrollers
#79 - Possibly useful for development

audiohacked, 2016-09-11:
I don't forsee any changes to this project or a fork because honestly there are not many people who understand how the microcontrollers work. These are not Atmel AVRs, or ARM7, this requires years of research before even writing a single line of code. Good Luck, I don't see you getting anywhere.

AshtinBlanchard, 2016-09-12:
@audiohacked Thank you for your support, your optimism is appreciated! Good luck to you to in all your adventures!

Junkya4, 2017-11-18:
Trying to do in SMI chips!