brandonlw/Psychson #155: Header not accepted

benji1000, 2016-09-16:
Hello,

I'm having an issue in the final step, when I try to flash the USB key with my custom firmware:

```
DriveCom.exe drive=F action=SendFirmware burner=BN03V117M.BIN firmware=FirmwareWithEmbeddedPayload.bin
Action specified: SendFirmware
Gathering information...
Reported chip type: 2303
Reported chip ID: 98-3A-94-93-76-51
Reported firmware version: 5.00.18
Mode: Firmware
Switching to boot mode...
FATAL: System.InvalidOperationException: Header not accepted
   ▒ DriveCom.PhisonDevice.TransferFile(Byte[] data, Byte header, Byte body) dans Q:\psychson\DriveCom\DriveCom\PhisonDevice.cs:ligne 256
   ▒ DriveCom.PhisonDevice.TransferFile(Byte[] data) dans Q:\psychson\DriveCom\DriveCom\PhisonDevice.cs:ligne 238
   ▒ DriveCom.Startup._ExecuteImage(String fileName) dans Q:\psychson\DriveCom\DriveCom\Startup.cs:ligne 403
   ▒ DriveCom.Startup._SendFirmware() dans Q:\psychson\DriveCom\DriveCom\Startup.cs:ligne 375
   ▒ DriveCom.Startup.Main(String[] args) dans Q:\psychson\DriveCom\DriveCom\Startup.cs:ligne 114
```

I saw many users having this problem because their USB key had the wrong controller version. However, I think my **Verbatim Store N Go V3** is compatible (see [info about my USB key](http://i.imgur.com/AzdwPKo.png)).

It has the right controller (2251-03). Also, I followed [this guide right here](http://null-byte.wonderhowto.com/how-to/make-your-own-bad-usb-0165419/). I've done it a few times just to be sure I wasn't making any ovious mistake. I'm using the same burner image as him (BN03V117M.BIN).

Could you help me figure out what I'm doing wrong? Thanks!

Apuru, 2016-09-25:
I have the same issue. I think it has something to do with the firmware. Most people are using firmware 1.XX.XX

We have 5.00.18

mdecaria, 2016-09-26:
Yep same here, get to last step and then run into this Header not accepted error...

I have FW version: 5.00.53 and its also a Verbatim Store N Go V3

mdecaria, 2016-09-26:
HA I got it working after many hours of testing and banging my head against the wall.

I used MPALL 3.72.0B with the following files:

Burner BN03V114M.BIN
Firmware FW03FF01V10810M.BIN

I got errors when flashing with MPALL but it still somehow downgraded the usb to 1.01.10

I used [https://www.pentestingshop.com/recover-a-usb-stick/](https://www.pentestingshop.com/recover-a-usb-stick/) as a guide for MPALL

Then ran the following:

`c:\fw\Psychson-master\tools\DriveCom.exe /drive=J /action=SendExecutable /burner="c:\fw\fw_bn\Firmware PS2251-03\BN03V114M.BIN"`

`c:\fw\Psychson-master\tools\DriveCom.exe /drive=J /action=SendFirmware /burner="c:\fw\fw_bn\Firmware PS2251-03\BN03V114M.BIN" /firmware=c:\fw\Psychson-master\firmware\bin\fw.bin`

Put the usb in another computer and off it runs my hello world payload :)

benji1000, 2016-09-27:
Unfortunately, I am unable to reproduce your procedure, @mdecaria. When plugging my USB stick, [I have error "Issue 2" in MPALL](http://i.imgur.com/iFA6Q6i.png). I've been trying a dozen times.

If anyone has any advice about this issue, please let me know. Otherwise, thank you @mdecaria, those for whom this solution works will be grateful to you :)

mdecaria, 2016-09-27:
Hey @benji1000

Looking at the attached image, did you put the usb in bootmode before using MPALL?
Did you have a look at that guide I linked? You might get errors when flashing but shouldn't get any when plugging it in when in bootmode...

benji1000, 2016-09-28:
Yes, I did read and use the guide you provided. I had a hard time prying open my USB key to access the electronic parts, and then shorting the pins while plugging it in. I've tried several times ot be sure the contact was made between the two pins. Here's [a picture of my USB key](http://i.imgur.com/ToQOtAK.jpg), with the dot on the micro-controller and the two pins I tried shorting.

mdecaria, 2016-09-28:
You should be shorting pins 2-3

Not the 1-2 pins.

benji1000, 2016-09-29:
Right, my mistake. Here's [an updated version](http://i.imgur.com/CAT7BR2.jpg). However, same result in MPALL.

I also tried using DriveCom from the wiki page you recommended, and that didn't work either. I have a different error though; instead of "Mode 255", I have "Mode 3", but same "ID Issue 0002".

mdecaria, 2016-09-29:
Setting boot mode through DriveCom won't work, the only way is by shorting those pins while plugging it in. Keep trying and then run `DriveCom.exe /drive=[driveLetter] /action=GetInfo`

It needs to say bootmode, if it's still in firmware, keep trying. There are a few videos out there showing everyone's own technique to short the pins

If you can't get it into bootmode, there isn't much else I can do to help, sorry

benji1000, 2016-09-29:
Well, here is something strange: I did what you told me, and even though MPALL still displays a red cell like there was an error, DriveCom reported this:

```
Reported chip type: 2303
Reported chip ID: 98-3A-94-93-76-51
Reported firmware version: 1.01.10
Mode: BootMode
```

Since the MPALL window didn't look like the one in the screenshot from the pentestingshop.com tutorial, I falsely assumed that I didn't execute the pins shorting correctly. The firmware is 1.01.10 because one time, I did however tried to follow the tutorial until the end.

So thank you for helping me and telling me to use DriveCom to get info about the key, seems to me that MPALL was misleading on this one!

So now that my USB was set to BootMode AND using a more appropriate firmware, I relaunched the first command, but was granted with an error message:

```
$ ../tools/DriveCom.exe drive=E action=SendFirmware burner=BN03V117M.BIN firmware=FirmwareWithEmbeddedPayload.bin
Action specified: SendFirmware
Gathering information...
Reported chip type: 2303
Reported chip ID: 98-3A-94-93-76-51
Reported firmware version: 1.01.10
Mode: BootMode
FATAL: System.InvalidOperationException: DeviceIoControl failed: 0079
   ▒ DriveCom.PhisonDevice._SendCommand(SafeFileHandle handle, Byte[] cmd, Byte[] data, Int32 bytesExpected) dans Q:\psychson\DriveCom\DriveCom\PhisonDevice.cs:ligne 365
   ▒ DriveCom.PhisonDevice.SendCommand(Byte[] cmd, Byte[] data) dans Q:\psychson\DriveCom\DriveCom\PhisonDevice.cs:ligne 314
   ▒ DriveCom.PhisonDevice.SendCommand(Byte[] cmd) dans Q:\psychson\DriveCom\DriveCom\PhisonDevice.cs:ligne 304
   ▒ DriveCom.PhisonDevice.JumpToPRAM() dans Q:\psychson\DriveCom\DriveCom\PhisonDevice.cs:ligne 228
   ▒ DriveCom.Startup._ExecuteImage(String fileName) dans Q:\psychson\DriveCom\DriveCom\Startup.cs:ligne 404
   ▒ DriveCom.Startup._SendFirmware() dans Q:\psychson\DriveCom\DriveCom\Startup.cs:ligne 375
   ▒ DriveCom.Startup.Main(String[] args) dans Q:\psychson\DriveCom\DriveCom\Startup.cs:ligne 114
```

This one is different from the first I had at the beginning of this thread, so I have to do some research on that, and then I guess that would be better if I started another issue.

mdecaria, 2016-09-30:
Did you run SendExecutable before running SendFirmware?

hugoreynoldsCwY, 2016-10-26:
@mdecaria 
SHA256(./BN03V114M.BIN)= 24b9e2b209a1b851fed3fdc32fe69255990befefec146ab3172c13f9b8a9b373
SHA256(./FW03FF01V10810M.BIN)= 8f511ed689dba81516a1c4b5ba4f7c8cddd08ad3483d48313e36d36ccc95a67c
Are these two files?