brandonlw/Psychson #157: Does the Phison PS2307 work?

ICallHax, 2016-10-01:
Hello,

So I decided to try my luck and buy a "Verbatim STORE N GO 32GB USB 3.0" which was said to be a compatible device. Upon checking it's controller I found that it was using a Phison PS2307 controller and was wondering if it is even possible for it to be converted to a BadUSB. I am also having trouble finding the burn image for it.

Here is a screenshot of the USB specs:
https://gyazo.com/4801c0a17f8f486cdfbfd0ce4222629a

Regards

mdecaria, 2016-10-01:
Phison PS2307 won't work and never will with the current code.

The best bet would be purchase a Verbatim STORE N GO V3 USB 3.0 8GB Drive (maybe the 32GB version) and look for the number under the barcode. If it starts with 83-P12 then it has a Phison 2251-03 controller

ICallHax, 2016-10-01:
@mdecaria Alright, thank you.

DominikGebhart, 2016-12-22:
I cannot confirm, mine reads _83-P120001869-2_ and contains a chip that reads _PS2251-07-V_