brandonlw/Psychson #159: This USB will work???

Darioso, 2016-10-05:
Hi, I have a USB with Phison 2251-68. It will work???

Volume: D:
Controller: Phison 2251-68
Possible Memory Chip(s): 
  Toshiba TH58NVG8T2JTA20
  Toshiba TH58TEG8T2JTA20
  Toshiba TH58TEG8T2JBA4C
  Toshiba TC58NVG7T2JTA00
  Toshiba TC58TEG7T2JTA00
Memory Type: TLC
Flash ID: 983AA892 7650
Chip F/W: 01.04.34
Firmware Date: 2015-03-11
VID: 18A5
PID: 0302
Manufacturer: Verbatim
Product: STORE N GO
Query Vendor ID: Verbatim
Query Product ID: STORE N GO      
Query Product Revision: 5.00
Physical Disk Capacity: 7740641280 Bytes
Windows Disk Capacity:  7732215808 Bytes
Internal Tags: 2P2P-SF5L
File System: FAT32
USB Version: 2.00
Declared Power: 100 mA
ContMeas ID: 033E-05-00
Microsoft Windows 10 x64 Build 14393
------------------------------------
http://www.antspec.com/usbflashinfo/
Program Version: 8.7.0.597

 Thanks

mdecaria, 2016-10-06:
No sorry, only Phison 2251-03 will work with the current code.

bitpage-de, 2018-03-22:
Is that up to date? Or is it possible to work with this usb stick in the meantime?