brandonlw/Psychson #162: Verbatim STORE N GO USB device

HackforumsRobot, 2016-10-10:
Having issues with it but it says it should work in the compatibility area. Number below the barcode is 83-P120004939-1
any help would be greatly appreciated.

c:\Users\Administrator\Videos\Psychson\tools>DriveCom.exe /drive=D /action=SendFirmware /burner="C:\Users\Administrator\Videos\firmware_ps2251-03\Firmware PS2251-03\BN03V114M.BIN" /firmware="C:\Users\Administrator\Videos\Psychson\firmware\bin\fw.bin"
Action specified: SendFirmware
Gathering information...
Reported chip type: 2268
Reported chip ID: 98-D7-84-93-72-57
Reported firmware version: 5.03.10
Mode: Firmware
Switching to boot mode...
FATAL: System.InvalidOperationException: DeviceIoControl failed: 0002
   at DriveCom.PhisonDevice._SendCommand(SafeFileHandle handle, Byte[] cmd, Byte[] data, Int32 bytesExpected) in C:\Users\Administrator\Source\Repos\Psychson\DriveCom\DriveCom\PhisonDevice.cs:line 362
   at DriveCom.PhisonDevice.SendCommand(Byte[] cmd, Byte[] data) in C:\Users\Administrator\Source\Repos\Psychson\DriveCom\DriveCom\PhisonDevice.cs:line 314
   at DriveCom.PhisonDevice.TransferFile(Byte[] data, Byte header, Byte body) in C:\Users\Administrator\Source\Repos\Psychson\DriveCom\DriveCom\PhisonDevice.cs:line 246
   at DriveCom.PhisonDevice.TransferFile(Byte[] data) in C:\Users\Administrator\Source\Repos\Psychson\DriveCom\DriveCom\PhisonDevice.cs:line 238
   at DriveCom.Startup._ExecuteImage(String fileName) in C:\Users\Administrator\Source\Repos\Psychson\DriveCom\DriveCom\Startup.cs:line 403
   at DriveCom.Startup._SendFirmware() in C:\Users\Administrator\Source\Repos\Psychson\DriveCom\DriveCom\Startup.cs:line 375
   at DriveCom.Startup.Main(String[] args) in C:\Users\Administrator\Source\Repos\Psychson\DriveCom\DriveCom\Startup.cs:line 114

mdecaria, 2016-10-10:
This usb won't work without quite a few extra steps.

This usb can only enter bootmode by short circuiting the pins (2-3)

Once in bootmode you can see the step I did here:
https://github.com/brandonlw/Psychson/issues/155#issuecomment-249545198

HackforumsRobot, 2016-10-10:
Hi there, this is the USB I am using - right now i don't wish to damage the case too much if I can, will this device work? 
The 8gb black
http://www.verbatim.com/prod/flash-memory/usb-drives/everyday-usb-drives/pinstripe-usb-drive-sku-49062/

mdecaria, 2016-10-10:
If you put the usb in bootmode and followed the steps, MPALL will spit out an error still which is OK.

Use this command is a to see if you successfully downgraded the firmware.

DriveCom.exe /drive=[driveLetter] /action=GetInfo

If it worked, then put the usb back into boot mode and run the other two commands in my tutorial post.

HackforumsRobot, 2016-10-10:
c:\Users\Administrator\Videos\Psychson\tools>DriveCom.exe /drive=D /action=GetInfo
Action specified: GetInfo
Gathering information...
Reported chip type: 2268
Reported chip ID: 98-D7-84-93-72-57
Reported firmware version: 5.03.10
Mode: Firmware

When I try to run it in MPALL I get CT-0x1042
http://i.imgur.com/hTqImGo.png

HackforumsRobot, 2016-10-10:
c:\Users\Administrator\Videos\Psychson\tools>DriveCom.exe /drive=D /action=GetInfo
Action specified: GetInfo
Gathering information...
Reported chip type: 2268
Reported chip ID: 98-D7-84-93-72-57
Reported firmware version: 5.03.10
Mode: Firmware

c:\Users\Administrator\Videos\Psychson\tools>DriveCom.exe /drive=D /action=GetInfo
Action specified: GetInfo
Gathering information...
Reported chip type: 2268
Reported chip ID: 98-D7-84-93-72-57
Reported firmware version: 1.02.10
Mode: BootMode

------------

I definitely made sure it was in BootMode, then ran MPALL which gave me an ISP error

mdecaria, 2016-10-10:
It's been downgraded, congratz

Now, you shouldn't need me anymore.
Just run the two commands on my other post and it should work

mdecaria, 2016-10-10:
I just realised that it's says you chip type isn't Phison 2251-03 (2303). So this whole thing won't work anyway.

You 100% sure you have a Verbatim STORE N GO V3 8GB

Does it say V3  ?

If you do then the compatibility list needs updating