brandonlw/Psychson #167: windows problem ??

soulremover, 2016-10-21:
hello there guys. I don;t own a Rubber Ducky, instead i found some usb sticks with the proper Phison chip and managed to "cook" a BadUsb. Ofc i am using the (latest) duckencoder.jar to make the inject.bin file. Everything worked out fine. The problem is that the payload is not running under Windows ... Tried both with the .jar and the python method. No matter what i try it's not working, (tried WINDOWS r, GUI r, CONTROL ESCAPE etc)
Here is the script i am using for the inject.bin file
DELAY 5000
WINDOWS r
DELAY 1000
STRING notepad
DELAY 1000
ENTER
DELAY 1000
STRING Hello World!!!
ENTER
When i plug in the usb nothing happens. When i boot with Kali Linux and open a terminal i can see it's working because words like "r" , "notepad" and "Hello World" are appearing 
Any help ? Thoughts on the matter ?
Thanks in advance !!

weasd10, 2016-10-23:
Today I tried using my USB that was able to exploit the BadUSB vulnerability successfully 6 months ago and got the same result you got.I think it may be the case that it has been patched in Windows 10,because it still works on other operating systems (Kali Linux and ChromeOS).If someone knows how to make it work on Windows please share some knowledge on how to do so.Thanks

mdecaria, 2016-10-23:
Works fine on my windows 10 box

I did run into the same problem however, it would work on most windows machine but there a few where it doesn't work at all.

soulremover, 2016-10-24:
are we able to find out the "why" ??

brandonlw, 2016-10-25:
I assure you, Windows 10 patched nothing because there is nothing to patch.
This is not an "exploit", it is an abuse of the nature of USB to pretend to
be other devices.

The problems you're experiencing may have to do with how the drive exposes
itself to the host, similar to how it behaves oddly on USB 3.0 ports but
not 2.0. It's a drive firmware issue.

Regardless, nothing has been "fixed" to address this in any meaningful way,
and It's doubtful it ever will be.

soulremover, 2016-10-26:
https://support.microsoft.com/en-us/kb/3143142

https://technet.microsoft.com/en-us/library/security/MS16-033

Any suggestion then @brandonlw  ?? any ideas ? what can i do to fix it ?

worst07, 2016-12-09:
i have the same problem