brandonlw/Psychson #168: Verbatim 32G FATAL: System.InvalidOperationException: Header not accepted

hugoreynoldsCwY, 2016-10-24:
Volume: E:
Controller: Phison 2303 (2251-03)
Possible Memory Chip(s): 
  Toshiba TH58NVG8T2JTA20
  Toshiba TH58TEG8T2JBA4C
  Toshiba TH58TEG8T2JTA20
  Toshiba TC58NVG7T2JTA00
  Toshiba TC58TEG7T2JTA00
Memory Type: TLC
Flash ID: 983AA892 76D7
Chip F/W: 01.01.10
Firmware Date: 2012-04-09
MP Ver.: MPALL vFF.FF.FF
VID: 13FE
PID: 5200
Manufacturer: Phison                  
Product: 2303 Boot ROM           
Query Vendor ID:         
Query Product ID: USB DISK 3.0    
Query Product Revision: PMAP
Physical Disk Capacity: 0 Bytes
Windows Disk Capacity:  0 Bytes
Internal Tags: 2R2F-ASK4
USB Version: 3.00 in 2.00 port
Declared Power: 498 mA
ContMeas ID: D296-13-00
Microsoft Windows 10 x64 Build 10240
------------------------------------
http://www.antspec.com/usbflashinfo/
Program Version: 8.7.0.599
C:\usb\Psychson-master\Psychson-master\tools>DriveCom.exe /drive=E /action=SetBootMode
Action specified: SetBootMode

C:\usb\Psychson-master\Psychson-master\tools>DriveCom.exe /drive=E /action=SendExecutable /burner=bn.bin
Action specified: SendExecutable

C:\usb\Psychson-master\Psychson-master\tools>DriveCom.exe /drive=E /action=SendFirmware /burner=bn.bin /firmware=fw.bin
Action specified: SendFirmware
Gathering information...
Reported chip type: 2303
Reported chip ID: 98-3A-A8-92-76-D7
Reported firmware version: 0.99.53
Mode: Firmware
Switching to boot mode...
Rebooting...
Sending firmware...
FATAL: System.InvalidOperationException: Header not accepted
   at DriveCom.PhisonDevice.TransferFile(Byte[] data, Byte header, Byte body) in C:\usb\Psychson-master\Psychson-master\DriveCom\DriveCom\PhisonDevice.cs:line 279
   at DriveCom.Startup._RunFirmware(String fileName) in C:\usb\Psychson-master\Psychson-master\DriveCom\DriveCom\Startup.cs:line 428
   at DriveCom.Startup.Main(String[] args) in C:\usb\Psychson-master\Psychson-master\DriveCom\DriveCom\Startup.cs:line 114

mdecaria, 2016-10-24:
I would try two things, either put your usb in bootmode manually by shorting circuiting the pins then trying again or..

Try reflashing the firmware with MPALL

Here's my post on how I got it working
https://github.com/brandonlw/Psychson/issues/155#issuecomment-249545198

hugoreynoldsCwY, 2016-10-26:
Thank you for your reply, I was destroyed by the controller. The new usb disk is on the way to mail