brandonlw/Psychson #173: Kingston DataTraveler G4 64 GB

ziziz, 2016-11-20:
Controller: Phison PS2307
Possible Memory Chip(s): Not available
Flash ID: 983A98A3 7651
Chip F/W: 01.02.55
Firmware Date: 2015-09-03
ID_BLK Ver.: 1.3.0.0
MP Ver.: MPALL v3.34.07
VID: 0951
PID: 1666
Manufacturer: Kingston
Product: DataTraveler 3.0
Query Vendor ID: Kingston
Query Product ID: DataTraveler 3.0
Query Product Revision: PMAP
Physical Disk Capacity: 62008590336 Bytes
Windows Disk Capacity:  61983457280 Bytes
Internal Tags: 2Q6P-8X74
File System: FAT32
Relative Offset: 4032 KB
USB Version: 3.00 in 2.00 port
Declared Power: 300 mA
ContMeas ID: 0620-06-00

peps96, 2017-02-23:
it is 2251- 07 aka PS2307, it cannot work with these files

sristisravan, 2017-03-08:
Any work around, suggested methods to make PS2307 device working as BadUSB?