brandonlw/Psychson #174: Id like to help research

zeroKilo, 2016-12-12:
hi,
I started by making your awesome PhisonDevice class into a GUI tool, so I can better debug whats happening. 
http://i.imgur.com/hPu1uXG.png 
But unfortunately not all commands work on my chip (2251-68-5) and I cant find suitable firmware, so Id rather like to research the ISP process itself more, especially the exchanged bytes, like, have you any idea what the header and body byte combo do, or is that just from traces? is there any documentation on the exchange other than the code itself? Is there a way I can contribute to this project with researching my chip variant or is that futile? I could help to document the protocol in the wiki if that helps. Thanks for this sw
greetz
WV

Jacoppy, 2016-12-15:
Personally I had the same idea,  I would like to update the project making it compatible with more modern firmware. 
I'd like to know the methodology used in order to achieve this and the tools used,  so I have an idea about the overall procedure. 
I'll wait some news from the owner of the repo.

zeroKilo, 2016-12-15:
well I can already flash newer firmwares and all, but I havent got to get my drive working again, but I learned a ton on the way. if you wanna know what uptool and mpall does, get wireshark with usbpcap and look for SCSI commands, followed by BULK in/out packets, and thats whats passed along. my tool even has a function now to send any byte sequence for testing, it simply "send some bytes, expect none/some bytes" and the class from this project already has SendCommand and the like, so its ready to use :)

well, if you want we can chat or so, but I dunno how to contact you

greetz

zeroKilo, 2016-12-22:
if you are still interested, I started to collect my research in a new forum here: http://phisonresearch.freeforums.net/thread/6/usb-communication-commands

greetz