brandonlw/Psychson #177: I don't think this will work.... KingstonDataTraveler 100G3 8GB

kobykotiv, 2017-02-10:
Volume: E:
Controller: Phison PS2307
Possible Memory Chip(s): Not available
Flash ID: 98DE88A3 7251
Chip F/W: 01.02.55
Firmware Date: 2015-09-03
ID_BLK Ver.: 1.3.0.0
MP Ver.: MPALL v3.34.0C
VID: 0951
PID: 1666
Manufacturer: Kingston
Product: DataTraveler 3.0
Query Vendor ID: Kingston
Query Product ID: DataTraveler 3.0
Query Product Revision: PMAP
Physical Disk Capacity: 7751073792 Bytes
Windows Disk Capacity:  7738507264 Bytes
Internal Tags: 2Q6P-8X74
File System: FAT32
Relative Offset: 4032 KB
USB Version: 3.00
Declared Power: 504 mA
ContMeas ID: 1B0D-06-00
Microsoft Windows 10 x64 Build 14393
------------------------------------
http://www.antspec.com/usbflashinfo/
Program Version: 8.8.0.600

sristisravan, 2017-03-02:
I got a Kingston Data Traveller 64GB, Below are the USB details, 
Controller: Phison PS2307
Possible Memory Chip(s): Not available
Flash ID: AD3A1803 0050
Chip F/W: 08.00.50
Firmware Date: 2016-09-01
ID_BLK Ver.: 1.4.6.0
MP Ver.: MPALL v5.35.12
VID: 0951
PID: 1666
Manufacturer: Kingston
Product: DataTraveler 3.0
Query Vendor ID: Kingston
Query Product ID: DataTraveler 3.0
Query Product Revision: PMAP
Physical Disk Capacity: 61958258688 Bytes
Windows Disk Capacity:  61907959808 Bytes
Internal Tags: 2Q6P-8762
File System: FAT32
Relative Offset: 30344 KB
USB Version: 3.00 in 2.00 port
Declared Power: 300 mA
ContMeas ID: D770-07-00
Microsoft Windows 7 SP1 x64 Build 7601
------------------------------------
http://www.antspec.com/usbflashinfo/
Program Version: 8.8.0.601

Its NOT WORKING!! Device control failed :0079, Any Suggestions or comments will be helpful
Thanks!

Meir1Bookatz, 2018-08-20:
you need the 2303 Chip, Phison 2251-03 microcontroller to work. i have the same kingstone traveler as you and it doesnt work.