brandonlw/Psychson #179: Trying to get a Cisco Easy Setup Key to work. Phison 2251-32 (2232)

BlueMustache, 2017-02-25:
Hey I have a fantastic little Cisco branded flashdrive that is a bit old, but I can't seem to get this to flash.
Here is the output. Please help. Thanks!

`Volume: E:
Controller: Phison 2251-32 (2232)
Possible Memory Chip(s): 
  Samsung K9G8G08U0B
Memory Type: MLC
Flash ID: ECD314A5 64
Chip F/W: 01.05.10
Firmware Date: 2008-01-26
VID: 13B1
PID: 002D
Manufacturer:         
Product: Easy Setup Key
Query Vendor ID:         
Query Product ID: Easy Setup Key  
Query Product Revision: PMAP
Physical Disk Capacity: 1003487232 Bytes
Windows Disk Capacity:  999354368 Bytes
Internal Tags: 2Q2P-SX9K
File System: NTFS
Relative Offset: 4032 KB
USB Version: 2.00
Declared Power: 200 mA
ContMeas ID: 7C0B-01-00
Microsoft Windows 8.1 x64 Build 9600`

cathome1, 2017-09-09:
Oh yeah. I also interested in this issue

cathome1, 2017-09-09:
but I have a memory chip:  Intel 29F16G08AAMC1
(silicon power)