brandonlw/Psychson #187: how to create badusb using Sandisk USB Help Me

killuakillu, 2017-05-24:
its tool used for sandisk yes or no  and help me for create badusb

grant-h, 2017-05-31:
Are you able to post the Vendor and Product IDs here?
Can you possibly crack open the flash drive and photograph the controllers chips inside so we can see what they are using?

killuakillu, 2017-06-03:
all information sandisk

Volume: G:
Controller: Alcor 0xF500
Possible Memory Chip(s): Not available
Flash ID: EC1A881F 70C8
Flash CE: 1
Chip F/W: FD00
Group: 98
VID: 058F
PID: 6387
Manufacturer: Generic
Product: Teclast CoolFlash
Query Vendor ID: Teclast 
Query Product ID: CoolFlash       
Query Product Revision: 8.07
Physical Disk Capacity: 15980298240 Bytes
Windows Disk Capacity:  15955156992 Bytes
Internal Tags: AYWR-S894
File System: FAT32
Relative Offset: 9312 KB
USB Version: 2.00
Declared Power: 200 mA
ContMeas ID: EED3-05-00
Microsoft Windows 10 x64 Build 14393
------------------------------------
http://www.antspec.com/usbflashinfo/
Program Version: 8.8.0.601

all infromation toshiba

Volume: G:
Controller: TC58NC6623 (SSS 6698) BA
Possible Memory Chip(s): 
  Toshiba TC58TEG6T2JTA00
  Toshiba TC58NVG6T2JTA00
Memory Type: TLC
Flash ID: 98DE9892 7257
Flash CE: 1
VID: 0930
PID: 6544
Manufacturer: TOSHIBA 
Product: TransMemory     
Query Vendor ID: TOSHIBA 
Query Product ID: TransMemory     
Query Product Revision: 1.00
Physical Disk Capacity: 7756087296 Bytes
Windows Disk Capacity:  7740706816 Bytes
Internal Tags: 396H-QAH4
File System: FAT32
Relative Offset: 31 KB
USB Version: 2.00
Declared Power: 200 mA
ContMeas ID: EED3-06-00
Microsoft Windows 10 x64 Build 14393
------------------------------------
http://www.antspec.com/usbflashinfo/
Program Version: 8.8.0.601

Junkya4, 2017-11-18:
At present, there is no firmware available.. so it can't be done.. try to find other drives ...

sai203, 2019-06-27:
1

sai203, 2019-06-27:
i am using sandisk 32gb,can it be turned into a bad usb
its details are here

Volume: H:
Controller: Silicon Motion SM3271 AB
Possible Memory Chip(s): Not available
Flash ID: 2CB47832
Flash CE: 1
Chip F/W: ISP 170308-DG2
MP: ::::2::
MPTOOL Ver.: 17.03.17
VID: 090C
PID: 1000
Manufacturer: SMI Corporation
Product: USB DISK
Query Vendor ID: SMI     
Query Product ID: USB DISK        
Query Product Revision: 1100
Physical Disk Capacity: 31750881280 Bytes
Windows Disk Capacity:  31742066688 Bytes
Internal Tags: QVAE-LSC2
File System: FAT32
Relative Offset: 1024 KB
USB Version: 2.00
Declared Power: 500 mA
ContMeas ID: 4BB4-05-EE
Microsoft Windows 7 SP1 x64 Build 7601
------------------------------------
http://www.antspec.com/usbflashinfo/
Program Version: 9.2.0.627

NicomoRuRu, 2022-03-11:
I have a 8GB cruzer blade and need to install a payload

NicomoRuRu, 2022-03-11:
Volume: E:
Controller: SanDisk
Possible Memory Chip(s): Not available
VID: 0781
PID: 5567
Manufacturer: SanDisk
Product: Cruzer Blade
Query Vendor ID: SanDisk 
Query Product ID: Cruzer Blade    
Query Product Revision: 1.26
Physical Disk Capacity: 8004304896 Bytes
Windows Disk Capacity:  7987511296 Bytes
Internal Tags: 3Y4F-8CB4
File System: FAT32
Relative Offset: 16 KB
USB Version: 2.00
Declared Power: 200 mA
ContMeas ID: 2B47-02-00
Microsoft Windows 10 x64 Build 19044
------------------------------------
http://www.antspec.com/usbflashinfo/
Program Version: 9.4.0.645