brandonlw/Psychson #198: Bricked Device and General Questions

jasonzhangwz, 2017-11-12:
I followed instructions online to make my own testing in order to train end users about unknow risk of BadUSB. However, I encountered several issues:
1) DUMP?
I think dump firmware equals to backup firmware from the usb. Am I right about this understanding?

2) Bricked Device
I prepared the following testing environment:
USB Device is San Disk, and control version is Phison 2251-03 (2303). (Used usbdriveinfo software to confirm this).
Installed Visual Studio Community 2015 x64 version and SDCC on my laptop.
Downloaded burner image BN03V104M.BIN from http://www.usbdev.ru/files/phison/
Compiled DriveCom.exe, Embedpayload and Injector without any errors.
Downloaded Payload Hello World code from https://github.com/hak5darren/USB-Rubber-Ducky/wiki/Payload---hello-world and save as a txt file name YouGotHacked.txt.
Then I used the following commands to burn my prepared image to my usb stick:
Dump Firmware:
tools\DriveCom.exe /drive=D /action=SetBootMode
tools\DriveCom.exe /drive=D /action=SendExecutable /burner=Psychson-master\firmware\BurnerImage_Fw2251_03\BN03V104M.BIN
tools\DriveCom.exe /drive=D /action=DumpFirmware /firmware=FW2203BKP.BIN
Prepare HID Payload:
java -jar Psychson-master\DuckEncoder\duckencoder.jar -i Psychson-master\DuckEncoder\YouGotHacked.txt -o Psychson-master\DuckEncoder\inject.bin
Go to folder \Psychson-master\firmware\ and run build.bat
tools\EmbedPayload.exe Psychson-master\DuckEncoder\inject.bin Psychson-master\firmware\bin\fw.bin (Prompt me file updated)
Send Payload to Firmware:
tools\DriveCom.exe /drive=D /action=SetBootMode
tools\DriveCom.exe /drive=D /action=SendExecutable /burner=Psychson-master\firmware\BurnerImage_Fw2251_03\BN03V104M.BIN
tools\Drivecom.exe /drive=D /action=SendFirmware /burner=Psychson-master\firmware\BurnerImage_Fw2251_03\BN03V104M.BIN /firmware=Psychson-master\firmware\bin\fw.bin
Then tragedy happened. It prompted me error but I didn't catch it.
After this, I tried many steps but failed to recover my usb stick. I cannot find it in computer management, and I used usbdriveinfo but couldn't find firmware.

This is the only usb device I found that can be used to make this test.

Can anyone help?

Thank you.

Junkya4, 2017-11-17:
Use Phison MPALL to recover..

jasonzhangwz, 2017-11-20:
Hi Junkya4:

Thanks for your information. I used three versions of Phison MPALL and also three versions of ChipGenius in order to find my device information and then recover it. However, all ChipGenius can identify nothing in the window, clicking "Update" in MPALL can also find nothing information so that I am not able to trigger to "start" recovery. Seems that I really bricked the device. My fault.
Will try to find some other device to give it a try.

Thanks again.