brandonlw/Psychson #199: How can this happen?

Junkya4, 2017-11-17:
Hello there,
As noted in the EmbedPayload program, it finds the bytes 0x12345678. Now seeing the bin image of the firmware file, this pattern is written after the command 0x22, which in 8051 is Return from function. My doubt is, how is the embedded payload in the firmware after the return statement, executed?

brandonlw, 2017-11-17:
The return statement you're seeing is likely from a subroutine. That or
another subroutine is accessing the address after the return statement in
order to read the payload and return it over USB in the form of keystrokes.

There's nothing suspicious about referencing data after code in 8051.

Junkya4, 2017-11-17:
Thanks for the reply in short time @brandonlw,
It's actually the end of the subroutine as it may be noticed. After the subroutine return, no function is trying to access the code beneath, as the bytes pattern is found and replaced.
Yes there's nothing wrong, but if the execution pointer never gets there, it couldn't be executed..!
So how do we say that it can definitely get executed?

brandonlw, 2017-11-17:
You seem to think the "payload" is 8051 code -- it isn't. It's a series of
bytes that the firmware reads from and interprets as a series of
keystrokes, which it then sends back over USB in HID format.

The execution pointer is never there. It's read from as data.

Junkya4, 2017-11-17:
Thanks again, got your point.
p.s.: off the topic, what can the statement 'firmware reads from and interprets as a series of keystrokes' possibly mean? I mean how the 'firmware' read and 'interpret'?

closed 2017-11-17 by Junkya4

brandonlw, 2017-11-17:
Rubber Ducky "payloads" are bytes that represent keys -- let's say, for
argument's sake, that 0x01 is "A", 0x02 is "B", etc. The flash drive
firmware reads a byte from the payload, interprets it as the letter "A",
then simulates the key down and key up event over USB, as a HID keyboard.

I'm not sure why you're confused over this -- it's the same thing a Rubber
Ducky's firmware is doing.

Junkya4, 2017-11-18:
Gotcha, still am confused, how will it be able to load, the low level details of course? It's like how will it interpret it's gonna (firmware) has to transfer over usb. Still confused, how would the firmware read the bytes. Like the firmware is loaded in the drive, the payload is inside the firmware too... Who reads the payload?
P.S.:- @brandonlw thanks for staying patient and answering.

brandonlw, 2017-11-18:
The firmware is capable of reading data embedded within itself. I don't
know what's confusing about that.

The firmware implements all high-level USB communication, including the USB
HID protocol used by keyboards and mice.
It can simulate keystrokes the exact same way a real keyboard would.

Junkya4, 2017-11-20:
Thank you, again.. Couldn't reply shortly. Thanks a lot.

brandonlw, 2017-12-04:
> Hello again @brandonlw <https://github.com/brandonlw>, is the default
> firmware itself capable of reading, or do we add this to it, as we are
> trying this over to a SMI device, it doesn't seem to be working. We Upload
> the custom firmware, in the file at multiple locations, the ISP checksum
> also changes but the payload isn't working.. Any ideas why not?

This obviously doesn't work on an SMI device. I don't know why you'd think
it would. Their firmware is completely different.

Your line of questioning suggests this is beyond your capabilities. I don't
think I can be of any more help.

Junkya4, 2017-12-06:
In response,
I think it is, and to extend them, I'm asking ur help with these questions... To be clear, SMI tool does send arbitrary commands to the chip. @brandonlw, I still can't understand which commands are run in the firmware to read itself,. AFAIK, there doesn't exist any software in which it has the code to read itself. Thanks for bearing.

brandonlw, 2017-12-06:
Now you're talking about developing firmware for a different controller
(SMI instead of Phison), which is beyond the scope of this repository.

With Phison, my earlier responses had to do with the 8051 code reading from
the running copy in RAM. Official firmware is much larger than the small
chunk running from RAM, and it has to read into RAM and execute the
sections it needs as it needs it. If you want to know how to read the
entire firmware from the NAND chip, I recommend you disassemble it and
study it.