brandonlw/Psychson #200: partial success with PS2251-07(2307) Verbatim StoreNGo 32Gb

sertyz, 2017-11-22:
I was partially success reflashing Verbatim StoreNGo 32GB based on PS2251-07(2307)  controller
using instructions and patches [https://vivibit.net/psychson2307final-en/](url)
Used pin short for entering boot mode. Hello World payload working, but I have issue reflasing to new payload. I’ve injected new payload into initially builded fw.bin, burning is ok, but I have still old payload. Is any untrivial procedure to write new fw.bin after initial reflashing procedure?

sflalex, 2018-02-17:
Hi sertyz, 
I have a different device (Kingstone Data Traveler G4 8GB) but same controller (PS2251-07).
I try with action=SendFirmare to send the saved original firmware from the device to test, but I have  Fatal System Invalid Operation Exception (Header not accepted), fails in line256,427, 378, 114.
I try with MPall and ST tool (a lot of idfferent versions) and allways the same error E255-1.
Can you explain me how do you send the Hello World Playload to your device?

Thanks for advanced.

p00n00b, 2019-07-03:
You have to modify burner.
https://github.com/p00n00b/2307-badusb/blob/master/README.md
I hope this repo to add some information about 2307 working devices.

four2ne, 2020-05-30:
> You have to modify burner.
> https://github.com/p00n00b/2307-badusb/blob/master/README.md
> I hope this repo to add some information about 2307 working devices.

i posted an issue about it on your repo, can u reply? thx