brandonlw/Psychson #202: Sandisk Ultra 16GB USB 3.0

arifsbc, 2018-03-27:
good sir, i have a question regarding bad usb with phison thing..
may you look at this link.. is this the right usb or not?
https://www.bukalapak.com/p/komputer/aksesoris-226/flashdisk/2sepjj-jual-sandisk-ultra-usb-3-0-100mb-s-flasdisk-16gb-cz48-distributor?blca=SESHO-KPFLD&gclid=Cj0KCQjw1-fVBRC3ARIsAIifYOOzf0RfGdKbeEq3dVYXgS4SuNPYHBIrxkvHLpiGBuPVjpDoSiAN8ikaAul8EALw_wcB

GeeDee-xd, 2018-05-21:
Yes. This is the correct USB. plus you can always refund or use it if it's not :)

Nima-Ra, 2019-08-09:
I got this USB, but it doesn't have Phison controller!
It has SanDisk instead and I suppose they've changed the controller so I think it should be removed from supported devices!
To me this project seems DEAD! There hasn't been any activity in a long time!
Is there any other choice?
Or any other fork of this project?