brandonlw/Psychson #206: "Unknown device" DT100 16GB bricked ?

SrMilton, 2018-06-12:
heyy boysssss. I have a Kingston DT100 16GB 3.0 (2307). After following the japanese guy tutorial, I did Hello World on my new badusb so after that I've tried to change de text. I coverted the code with duckenconder and did the same thing of original hello world and when I connect my USB on the port i get the unknown device error. I've tried to pin 2 and 3 to bootmode but still the same error. There is a way to repair this ?
[image]

nunofaria007, 2018-09-09:
Short pins 26 and 27 to enter BootMode

[image]

SrMilton, 2018-09-12:
Sorry, 2 and 3 = 26 and 27. Still the same ;/

nunofaria007, 2018-09-12:
Strange, i have a 2307 and no matter how many times i brick it i can allways enter boot mode by shorting the pins.

nunofaria007, 2018-09-12:
Just to be sure, are you shorting the pins as you plug the device, or are you trying to short them after plugging?

SrMilton, 2018-09-12:
I'm shorting the pins and after I connect the device with the pins shorted

SrMilton, 2018-09-12:
Is it common notify me as "Unknown device" but works fine on Drivecom ?

nunofaria007, 2018-09-12:
Yes, it happened the same to me, if you go to Device Manager and change the drivers from Composite to Mass Storage you'll see that the device is recognized with no problems. I guess there is some trouble with the descriptors. I managed to get this to work after removing the Mass Storage Device interface and it's endpoints from the descriptor in the source code, so the device will be recognized only as a keyboard and not as a composite device. The downside is that i lost the ability to send SCSI commands (now i need to short the pins every time i want to jump to BootMode). In your case, if you can communicate with the device using DriveCom, then you can put it in BootMode and use MPALL to burn the official firmware.

closed 2020-08-24 by SrMilton