brandonlw/Psychson #207: FATAL Problem

konstantinosskoutelis, 2018-09-11:
I tried to execute everything as you describe and in the end trying to flash the firmware as such << "PATH to DriveCom.exe" /drive="Drive Letter" /action=SendFirmware /burner="PATH to our burner image" /firmware="PATH to the firmware" >> I get the error showing below. What could be the problem? Thank you in advance

```
FATAL: System.InvalidOperationException: Header not accepted
   at DriveCom.PhisonDevice.TransferFile(Byte[] data, Byte header, Byte body) in C:\Users\skout\Desktop\RUBBER DUCKY\Psychson-master\DriveCom\DriveCom\PhisonDevice.cs:line 256
   at DriveCom.PhisonDevice.TransferFile(Byte[] data) in C:\Users\skout\Desktop\RUBBER DUCKY\Psychson-master\DriveCom\DriveCom\PhisonDevice.cs:line 238
   at DriveCom.Startup._ExecuteImage(String fileName) in C:\Users\skout\Desktop\RUBBER DUCKY\Psychson-master\DriveCom\DriveCom\Startup.cs:line 403
   at DriveCom.Startup._SendFirmware() in C:\Users\skout\Desktop\RUBBER DUCKY\Psychson-master\DriveCom\DriveCom\Startup.cs:line 375
   at DriveCom.Startup.Main(String[] args) in C:\Users\skout\Desktop\RUBBER DUCKY\Psychson-master\DriveCom\DriveCom\Startup.cs:line 114
```

nunofaria007, 2018-09-12:
I think that happens when the Burner is being sent and the device isn't on BootMode. I believe some official firmwares ignore the BF command. To check this:

- Run DriveCom with /drive="Drive Letter" argument

- Write "boot"

- Write "mode"

If it reports Firmware mode it means you can't jump to BootMode using DriveCom. If this is your case then you need to enter BootMode manually by shorting the pins (depends on the version of your micro-controller).

konstantinosskoutelis, 2018-09-12:
Thank you for the answer. I tried it but it returns this `Gathering information...
Reported chip type: 0000
Reported chip ID: 20-AE-00-01-00-00
Reported firmware version: 111.6E.66
Mode: Unknown` .Is this Firmware mode you are talking about ?

nunofaria007, 2018-09-12:
Which pendrive do you have? I think yours might not have a Phison micro-controller.

konstantinosskoutelis, 2018-09-12:
Well I got mine from the Known Supported Devices https://github.com/brandonlw/Psychson/wiki/Known-Supported-Devices and it is the Sandisk Ultra 16GB USB 3.0

nunofaria007, 2018-09-12:
Replug your pendrive and repeat the steps but this time don't write "boot". Let's see if this time it report the chip type.

konstantinosskoutelis, 2018-09-12:
It's still the same result. I tried multiple times the same steps but nothing changed and that's why it bothered me the most

nunofaria007, 2018-09-12:
There is a software called ChipGenius that might give some intel on your device.

konstantinosskoutelis, 2018-09-12:
Probably needs Chinese Language or something cause the characters do not show up

nunofaria007, 2018-09-12:
Can you try on a more recent version of ChipGenius? That one is indeed not written in english.

nunofaria007, 2018-09-12:
Btw, your pendrive looks like it's plugged on an HUB. Can you plug it directly on an USB port?

konstantinosskoutelis, 2018-09-12:
[image]
You were right

nunofaria007, 2018-09-12:
Now it shows up as a Mass Storage Device and with the correct vendor ID. Was it plugged on an HUB?

konstantinosskoutelis, 2018-09-12:
No it's not

nunofaria007, 2018-09-12:
Looks like ChipGenius can't retrieve info about the controller. Can you still use your pendrive as normal or is it bricked?

konstantinosskoutelis, 2018-09-12:
Yes I can use it normally

nunofaria007, 2018-09-12:
Well, you can try the GetInfo utility that comes with MPALL. That should be able to retrieve the info about your device's controller.

SrMilton, 2018-09-27:
Yeh, I've disassembled a Sandisk Ultra 16GB USB 3.0 and I got this shit. idk if it works
[image]
[image]
[image]

konstantinosskoutelis, 2018-09-28:
Then it should definitely be removed from the list with the suggested USB drives to prevent any other misunderstandings

nunofaria007, 2018-09-28:
The problem is that after some time, brands start to use different usb controllers in the production of the same product. In the past Sandisk Ultra 16GB USB 3.0 might have been produced using Phison controllers, and now they might be produced with different components.

SrMilton, 2018-09-28:
Yep, I'm trying to sell this pendrive to buy another one... Please, update the list :'(

nunofaria007, 2018-09-28:
Higly doubt that someone will update it... This project seems dead now.

SrMilton, 2018-09-28:
Yeh, I know.. Do you think is better do a badusb or pay $45 in a rubberducky ? If badusb >>, there is some fork better than the original project ?

nunofaria007, 2018-09-28:
I came across with this project and played a bit with it on my Phison 2307. To my knowledge there aren't any forks (at least anything worth mentioning). If you really need this that bad, then a rubberducky might be your best option (but i think 45$ is waaaaaay overpriced, specially if you want to play a prank on someone). Keep in mind that when you burn badusb firmware to your pendrive it will act as a keyboard only and you will be unable to save files on it unless you return to the original firmware (you will also need to reburn the badusb firmware whenever you want to change the payload). On that matter rubberducky has an advantage because it has and embedded sdcard slot (according to their website)..

SrMilton, 2018-09-30:
Tks, I'll buy a Malduino Elite. 16 payloads in the same time. Price almost the same