brandonlw/Psychson #213: Does a ps2309 controller supported ?

lessar69, 2019-07-12:
Volume: E:
Controller: Phison PS2309
Possible Memory Chip(s): Not available
Flash ID: 983C98B3 7672
Chip F/W: 08.05.5D
Firmware Date: 2018-03-01
ID_BLK Ver.: 1.4.33.0
MP Ver.: MPALL v5.35.35
VID: 0951
PID: 1666
Manufacturer: Kingston
Product: DataTraveler 3.0
Query Vendor ID: Kingston
Query Product ID: DataTraveler 3.0
Query Product Revision: PMAP
Physical Disk Capacity: 62008590336 Bytes
Windows Disk Capacity:  61958455296 Bytes
Internal Tags: 2Q6P-8X74
File System: FAT32
Relative Offset: 30336 KB
USB Version: 3.00 in 2.00 port
Declared Power: 300 mA
ContMeas ID: FC8E-12-00

enygaseluce, 2022-01-01:
i have the same problem and the same usb

enygaseluce, 2022-01-01:
Volume: D:
Controller: Phison PS2309
Possible Memory Chip(s): Not available
Flash ID: 2CA40832 A1
Chip F/W: 08.03.5D
Firmware Date: 2018-05-24
ID_BLK Ver.: 1.4.36.0
MP Ver.: MPALL v5.35.36
VID: 0951
PID: 1666
Manufacturer: Kingston
Product: DataTraveler 3.0
Query Vendor ID: Kingston
Query Product ID: DataTraveler 3.0
Query Product Revision: PMAP
Physical Disk Capacity: 31029460992 Bytes
Windows Disk Capacity:  31006867456 Bytes
Internal Tags: 2Q6P-8X74
File System: FAT32
USB Version: 3.10
Declared Power: 504 mA
ContMeas ID: C8BC-06-00

Ikram2277, 2022-03-08:
64GB USB not partition create unlocated disk space 57.62 GB
Description: [G:]USB Mass Storage Device(Kingston DataTraveler 3.0)
Device Type:  Mass Storage Device

Protocal Version: USB 2.10 <- Hint: This device can run faster when plugged to a USB3.0 port
Current Speed: High Speed
Max Current: 300mA

USB Device ID: VID = 0951 PID = 1666
Serial Number: E0D55E6CE772F4C1691825B9

Device Vendor: Kingston
Device Name: DataTraveler 3.0
Device Revision: 0001

Manufacturer: Kingston
Product Model: DataTraveler 3.0

Controller Part-Number: Unknown